cbm80amiga/Arduino_ST7789_Fast #1: Errors compiling for nodemcu

Bert-Tec, 2020-03-30:
Hi,

I get the following error when I try to compile the code with your library for nodemcu (ESP-12).
It works fine with the Adafruit library.

```
#include <SPI.h>
#include <Adafruit_GFX.h>    // Core graphics library
//#include <Adafruit_ST7789.h> // Hardware-specific library for ST7789
#include <Arduino_ST7789_Fast.h>

// ST7789 TFT module connections
#define TFT_DC    5     // TFT DC  pin is connected to NodeMCU pin D1 (GPIO5)
#define TFT_RST   4     // TFT RST pin is connected to NodeMCU pin D2 (GPIO4)
//#define TFT_CS    D8     // TFT CS  pin is connected to NodeMCU pin D8 (GPIO15)

#define SCR_WD   240
#define SCR_HT   240 

#define COMPATIBILITY_MODE
//#define CS_ALWAYS_LOW

//Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_RST);
Arduino_ST7789 tft = Arduino_ST7789(TFT_DC, TFT_RST);
```

```
C:\---\Arduino\libraries\Arduino_ST7789_Fast\Arduino_ST7789_Fast.cpp:62:0: warning: "CS_IDLE" redefined [enabled by default]

 #define CS_IDLE

 ^

C:\---\Arduino\libraries\Arduino_ST7789_Fast\Arduino_ST7789_Fast.cpp:56:0: note: this is the location of the previous definition

 #define CS_IDLE    *csPort |= csMask

 ^

C:\---\Arduino\libraries\Arduino_ST7789_Fast\Arduino_ST7789_Fast.cpp:63:0: warning: "CS_ACTIVE" redefined [enabled by default]

 #define CS_ACTIVE

 ^

C:\---\Arduino\libraries\Arduino_ST7789_Fast\Arduino_ST7789_Fast.cpp:57:0: note: this is the location of the previous definition

 #define CS_ACTIVE  *csPort &= ~csMask

 ^

C:\---\Arduino\libraries\Arduino_ST7789_Fast\Arduino_ST7789_Fast.cpp: In member function 'void Arduino_ST7789::writeSPI(uint8_t)':

C:\---\Arduino\libraries\Arduino_ST7789_Fast\Arduino_ST7789_Fast.cpp:75:5: error: 'SPDR' was not declared in this scope

     SPDR = c;

     ^

C:\---\Arduino\libraries\Arduino_ST7789_Fast\Arduino_ST7789_Fast.cpp: In member function 'void Arduino_ST7789::commonST7789Init(const uint8_t*)':

C:\---\Arduino\libraries\Arduino_ST7789_Fast\Arduino_ST7789_Fast.cpp:169:10: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment

   dcPort = portOutputRegister(digitalPinToPort(dcPin));

          ^
```

Can you help me?
As far as I found by asking google it's something about the SPI.

cbm80amiga, 2020-03-31:
This library is optimized for AVR so will not work with ESP8266.
Regards,
Pawel

gotnull, 2021-01-08:
Is there any chance you could optimize it for the ESP8266?

bpieres, 2021-01-25:
Same problem :(

wwitesp, 2021-04-22:
I've rolled back to an earlier build dated somewhere in 2019 and hashed out the //#include <avr/pgmspace.h> in the Arduino_ST7789_Fast.h file and am working again. I'm using an ESP32 based board and this lib works great.

wwitesp, 2021-04-22:
> […](#)

*** Apologies - this should not be here and cannot delete this ***