cbm80amiga/Arduino_ST7789_Fast #3: Use with 320x240 TFT...

Kentola, 2020-08-30:
Thanks for a nice library...

I have a ST7789 TFT 320x240 and I can get it to work with some of the examples, but the rest of screen outside 240x240 is flickering as assume it should be in this mode...

I’m not an experienced Arduino programmer, but I’ve tried to change to 320x240 in the example code, but I am not sure exactly where to set it up, if possible...?

Is there any chance to update or change code in this driver for use 320x240 TFT screens...?

https://www.waveshare.com/wiki/2inch_LCD_Module

cbm80amiga, 2020-09-01:
Hi,
Sorry I don't have such version of ST7789 display. You can try to define
frame buffer limits like in scroller example, maybe it will work.
Regards,
Pawel

Kentola, 2020-09-01:
Ok

Thanks! I’ll try that...

Brgds
Kent Tillaeus

ht93, 2021-05-09:
> Ok Thanks! I’ll try that... Brgds Kent Tillaeus
> […](#)

Hi, Have you solve the issue? I have the same display and want to try this library. Thanks a lot!

sbhw, 2022-02-17:
Arduino_ST7789::fillRect() will overflow in 240x320 setting because 240x320 exceeds unsigned 16-bit integer limit. Might need to call writeMulti() multiple times or modify the Arduino_ST7789::writeMulti() to uint32_t type.