cbm80amiga/Arduino_ST7789_Fast #9: Error compiling for SAMD Board

ludovicb1239, 2021-04-03:
I'm using arduino mkr wifi 1010 and I'm getting this error:
Arduino\libraries\Arduino_ST7789_Fast-master/Arduino_ST7789_Fast.h:19:10: fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>

Will this ever work for a SAMD board or am I just missing a library ?

cbm80amiga, 2021-04-04:
This library is optimized for AVR, never tested it with other MCUs

SCWhite, 2021-11-25:
HI @ludovicb1239  I manage to use this library on my ArduinoZero successfully 

Arduino mkr wifi 1010 and ArduinoZero are actually using the same mcu (SAMD21).

so I think it should work, you just need to enable the "COMPATIBILITY_MODE" in the Arduino_ST7789_Fast.h file
I didn't have the same error you're getting, maybe some lib is missing?

let me know if you still have compiling problem, and we can work it out.