cbm80amiga/Arduino_ST7789_Fast #11: blurry / not working  / resister needed with Arduino UNO

SCWhite, 2021-11-22:
HI, this is really a good library, thank you for your contribution.

I'm testing this library on an Arduino UNO board just as the YouTube video shows: [LINK](https://www.youtube.com/watch?v=GciLKcWQZK4) 
connecting my IPS 240x240 display directly on UNO 

but my display are always "black" (or just very blurry with only some pixel moving)
I only get the correct image in a split second  when I press the reset button.

After searching for a while, I see a comment under that YouTube video saying adding some 1K resister will help.

so I have tested it and can confirm adding an 1K resister on each communicate pin (SCA/SCL/RES/DC) will resolve this problem.

I'm wondering this might be a "hardware issue" but not a "software issue",  might have something to do with 5V instead of 3.3V system
so maybe we can add some instruction / notice on "README.md" or even in the file to help future user ?

if the author thinks this is a thing worth to do, I can test most boards on my hands and create a pull request.

Any suggestion will be appreciate, or may be I can be wrong on the hole thing(?)

cbm80amiga, 2021-11-22:
Use regular 3V3, don't use any resistors because they add extra noise.
Make sure that BLK has the same function like described here (try to connect it to GND or VCC)

SCWhite, 2021-11-25:
After a few days, I think I can describe this problem properly.

the blurry / black display issue is caused by using different SPI logic level on different boards.

the IPS 240x240 we're using accept 3.3V level communication, while some dev-boards have 5V
in that video, the author use a arduino mini, which is a 3.3v level SPI  

It's pretty clear on arduino's official reference:
https://www.arduino.cc/en/reference/SPI

[image]

so adding an 1K resistor make it work on UNO.

I really think we can add this information on  "README.md" to create a better quality of life when using this library.

BTW I manage to use this library on my ArduinoZero(with COMPATIBILITY_MODE on)
by connecting SDA -> ICSP-4 / SCL -> ICSP-1 / RES -> D8 / DC -> D7