cbm80amiga/Arduino_ST7789_Fast #15: Dynamic text

waw555, 2022-04-05:
Hello!

How to make dynamic text.

There is a text and a value after the text, how to make the value not overlap each other?

`   
void setup(){
   while (1) {
      sens_signal = analogRead(IR_SENSOR_PIN);
      lcd.setCursor(10, 30);
      lcd.setTextColor(RED);
      lcd.setTextSize(2);
      lcd.println("IR");
      lcd.setCursor(40, 65);
      lcd.println("SENSOR");
      lcd.setCursor(100, 90);
      lcd.println(sens_signal);
      if (millis() - reset_timer > 5000) break;
    }
}`

I need to display a digital value from the sens_signal variable, it changes and overlaps, it turns out nonsense.

lcd.println(sens_signal);

cbm80amiga, 2022-04-06:
Try to use lcd.setTextColor(RED,BLACK) to clear text background (like in Adafruit lib).

waw555, 2022-04-06:
Pavel, thank you very much, I will try.

Right now I have to display text in black before outputting new text, which is not very convenient when used in a loop, very strong screen flicker.

closed 2022-04-06 by waw555